GregDMeyer/IT8951 #2: memory access error

SirDagen, 2019-11-29:
I try using it with an 7.8" display ( https://www.waveshare.com/wiki/7.8inch_e-Paper_HAT ) but I onl get:

Initializing EPD...
Speicherzugriffsfehler (*which is: memory access error*)

P.S. I also installed you PaperTTY ( https://github.com/GregDMeyer/PaperTTY ), and get my shell displayed correctly, but the following error:

Initializing...
setting TTY size to 70 rows, 156 cols
Could not set TTY size (rows=70, cols=156); continuing.
Running...

When I CTRL+C I also get the sleeping penguin. So at least something is working (hopefully this additional information helps).

SirDagen, 2019-11-29:
Jesus Christ, it runs with "sudo python test.py"!

How can I get it running without?

GregDMeyer, 2019-11-29:
Great! Unfortunately you need sudo to run this, because the SPI communication with the device requires root (as far as I know). However it would be good to add a check so that instead of just a seg fault you actually get a useful error message. I'll do that!

closed 2019-11-29 by GregDMeyer

SirDagen, 2019-11-29:
Not sure. The other Python drivers (who also work with SPI) dont need them. I have an older 4.2" display from them:

https://github.com/waveshare/e-Paper

However, the main thing is that it works. Thank you, Greg

BackSlasher, 2019-12-01:
I suspect it depends on ownership for the SPI device:
```
$ ls /dev/spidev0.0 -l
crw------- 1 root root 153, 0 Nov 21 02:29 /dev/spidev0.0
```
Try `sudo chown $USER /dev/spidev0.0` (will be reverted when you restart your device) to see if that's the problem

GregDMeyer, 2019-12-01:
Ah---for this library I am not going through `/dev/spidev*` because I was running into some problems (perhaps a kernel bug) that were not appropriately toggling the CS pin during data transmission. Instead, I'm calling the bcm2835 library and doing the SPI communication directly through that. Hopefully there is a way to do through `/dev/spi*`, I'll look into that when I have a chance!

SirDagen, 2019-12-02:
Well I also had to install bcm2835-1.60 for the older 4.2" ePaper where this link is from. Thats all I know