GregDMeyer/IT8951 #4: RFC: Split GPIO-specific parts into methods, so they can be overriden in inheritence

BackSlasher, 2019-12-01:
Thank you very much for writing this module! I've been failing to do this for a couple of months and was delighted to find it.

I'm planning a project where a single RPi would drive multiple Waveshare screens (10+), and to avoid running out of GPIO pins I'm planning to:
1. Use the same SPI port for all of them
2. Use different solutions for the "unique pins" (HRDY, CS, RESET). My current idea is extending them using a USB-connected Arduino, but I can also use some sort of demuxer.

In any case, I'll need to customize the methods that drive those pins.
My utopian vision would be:
1. Moving all GPIO-specific operations to specific methods
2. Subclassing the Interface / SPI classes, replacing those methods with my own implementation
3. Subclassing the AutoDisplay class to use my version of Interface / SPI

I'll be happy to put up a PR, but I wanted to know your opinion on moving those GPIO operations to separate methods.
You can see my current (yet untested!) attempts in https://github.com/BackSlasher/IT8951-1/tree/inheritable-gpio
Thank you for your time!

SirDagen, 2019-12-19:
Sounds interesting.

GregDMeyer, 2020-06-07:
going to close this for now since it seems like there isn't a push to get this working any more, but if anyone is ever looking for this in the future, we can reopen!

closed 2020-06-07 by GregDMeyer