GregDMeyer/IT8951 #7: Segmentation fault on raspberry pi 4

ariep, 2020-04-19:
When trying to use this library on a raspberry pi 4, I get the following error:
```
bcm2835_init: gpio mmap failed: Cannot allocate memory
Segmentation fault
```
This is with version 1.65 of the bcm2835 library (which includes fixes to support the raspberry pi 4), and running as root.

Also, the bcm2835 test program included in that library runs correctly.

Could it somehow be due to the way bcm2835 is used?

GregDMeyer, 2020-04-22:
oh bummer! OK, thank you for the info. I don't have a RPi4 to test on, but I can still do some investigation and see what I find.

By the way, the only reason I was using the bcm2835 library in the first place was because I was encountering some strange behavior with /dev/spidev* through the spidev python module. I will revisit that; if it has been fixed now then we might not need the C library at all! (and probably not need sudo either!)

ariep, 2020-04-22:
This problem disappeared after I recreated the python virtualenv that I installed this library in.

I probably installed IT8951 before I updated the bcm2835 dependency to a version that has rpi4 support. Even after updating bcm2835, the problem persisted, perhaps because the python binding to it had incorporated constants from the C library's header file into its compiled source, just guessing here. Anyway, I don't get the error anymore.

> I don't have a RPi4 to test on, but I can still do some investigation and see what I find.

Feel free to let me know if you have any changes that you want me to try on my rpi4.

Currently I do have another problem though: the bcm2835 init function just hangs. Shall I open a new issue for that here?

closed 2020-04-22 by ariep

GregDMeyer, 2020-04-24:
ok, cool! Today I am exploring if I can make some changes to how the communication is handled---hopefully that can fix whatever is causing your hang, too. If you want to open an issue, though, please do! we can hopefully close it if the new changes fix things :)

GregDMeyer, 2020-05-18:
Hey @ariep --- heads up I just finished rewriting the backend. Would you mind testing it on RPi 4? Hopefully everything should work now! BTW, I'm about to push a change for your other issue :)