GregDMeyer/IT8951 #9: Initializing EPD

rgodragon, 2020-05-24:
Thanks for sharing the amazing code.
I tried to control the display with this code, but I failed.
I don't know what the problem is, so I would like to seek your advice.

After downloading the repository, I ran three commands.
```
1) pip install -r requirements.txt
2) pip install ./
3) Python setup.py build_ext —inlace
```

It turned out that they were all installed successfully.

And then, when I run ‘test.py’,
The following screen is displayed.

> Initializing EPD...
> VCOM set to 0.0
> System info:
>   display size: 0x0
>   img buffer address: 0
>   firmware version: 
>   LUT version: 
> 
> Clearing display...
> Traceback (most recent call last):
>   File "test.py", line 51, in <module>
>     main()
>   File "test.py", line 45, in main
>     t(display)
>   File "/home/pi/Desktop/kr/IT8951-master/test/integration/test_functions.py", line 38, in clear_display
>     display.clear()
>   File "/home/pi/Desktop/kr/myvenv/lib/python3.7/site-packages/IT8951/display.py", line 112, in clear
>     self.draw_full(DisplayModes.INIT)
>   File "/home/pi/Desktop/kr/myvenv/lib/python3.7/site-packages/IT8951/display.py", line 54, in draw_full
>     self.update(self._get_frame_buf().tobytes(), (0,0), (self.width, self.height), mode)
>   File "/home/pi/Desktop/kr/myvenv/lib/python3.7/site-packages/PIL/Image.py", line 733, in tobytes
>     e.setimage(self.im)
> SystemError: tile cannot extend outside image

As you can see, it doesn’t get the width and height of the display, img_buf_address, firmware_version and lut_version information.
It seems that the above error caused PIL error too (tile cannot extend outside image).

When I ran the c code which was provided by waveshare,
(https://www.waveshare.com/wiki/6inch_HD_e-Paper_HAT)
The display worked well. So I don't think it's not a hardware issue.

The display I got has the Vcom value ‘-2.09’,
I changed all the Vcom value in the _display.py / interface.py / test.py / test_xfer.py / time_full.py / time_partial.py_
If this is a problem, please let me know what parts should be changed the Vcom value.

I am a beginner with Raspberry Pi, Python, and E-ink Paper.
I would appreciate any advice on what the problem is and how to solve it.

GregDMeyer, 2020-05-24:
Ah, too bad! Thanks for letting me know about this issue.

When the device gives all zeros like that, it's because the communication with the device failed. I would try perhaps reducing the SPI transfer frequency---maybe it is too fast for your device to handle, for some reason. You can go into the source code and change it. Probably the easiest way is by passing "max_speed_hz=1000000" to the SPI class in line 24 of `interface.py`. Hopefully today I will upload some changes that make that value a lot easier to modify directly from the test code, so you could also wait for that.

By the way, `python setup.py build_ext —inplace"`shouldn't be necessary. The extension should automatically build when you do `pip install ./`. If you are getting an `ImportError` on `spi`, that might be because you were testing it from the top-level directory of the repository (where `setup.py` is) and it was looking at the local files first, instead of the installed ones. If you `cd` to another directory (even the test directory) you should not get an ImportError.

Let me know if it fixes things!

rgodragon, 2020-05-25:
Hi. Thanks for your kind reply.
I noticed that this could be a problem with SPI speed or SPI installation.

As you recommended, I proceeded only `pip install ./`  except for `python setup.py build_ext —inplace `.
Also, I lowered the ‘`max_speed_hz = 8000000-> 1000000`’ (in line 48 of `spi.pyx`) and then ran the code (I'm not sure this was what you mean)

But still, the device gives all zeros.

So I connected the new 6-inch display I ordered together and ran it with the same code.
Then, surprisingly, it worked!
I could see the penguin lying down ..!
Below is the screen I saw.

> Initializing EPD ...
> VCOM set to -1.5
> System info:
>   display size: 800x600
>   img buffer address: 118D30
>   firmware version: WS_v.0.1
>   LUT version: 6M14T
> 
> Clearing display ...
> Displaying gradient ...
> Starting partial update ...
>   writing full ...
>   writing partial ...
> Displaying "images / sleeping_penguin.png" ...
> Done!

Now, what I've found is that the code doesn’t work only for the display I used yesterday.
It works fine through c code (from waveshare), so it is not considered a hardware flaw.
As you suggested, if you modify the code to easily control the SPI speed, I will test the problematic display again :)
I am looking forward to updated code. 

Thank you again for sharing code and fast reply!

closed 2020-06-01 by rgodragon

GregDMeyer, 2020-06-02:
sorry to not get back to you last week on this---after you modified spi.pyx, did you change the line in setup.py to `USE_CYTHON=True`? Otherwise it will not recompile spi.pyx. Let me know if it helps! (You can also pass the SPI frequency directly to the SPI class in `interface.py` and it will override the default in `spi.pyx`).