GregDMeyer/IT8951 #10: Thank you!

SirDagen, 2020-05-30:
Thank you Greg, this software is wonderful. I tested the current update on my "Raspberry Pi 3 Model A+" with a "Waveshare 9.7inch e-Paper HAT" and it works wonderfully. The data transfer is so much faster now. Amazing. 

I build a sun clock that displays the current position and trajectory of the sun on moon and gifted it a new large display. Love the large ePaper stuff. So fun to use.

https://lh3.googleusercontent.com/pw/ACtC-3f7CD4Wt5vcILuu1VbvlI8I9WukWYERkiTjWHC5DuKm0xW0lxZ8gj3yr68x75-m_zpwwW6YljhObwkDroK2J5bDM4WOdGqZbQUCzfuPcxxeVwh6gEbvh58XMCjD-0SpalyHfHpPGjYkGeepgbyrs8pi=w981-h1307-no?authuser=0

closed 2020-05-30 by SirDagen

GregDMeyer, 2020-06-02:
Wow, this was a really sweet message! Thanks! I'm glad you are getting good use out of it :)

GregDMeyer, 2020-06-02:
also, that is a really cool application for it! have you posted the code somewhere?

SirDagen, 2020-06-03:
It came from my heart and I honestly love what you did.
 
Thank you. The code is not yet published because it is the first time I code on a Raspberry, the first time I use Python and the first time I use an ePaper display. Thus the application code is incredibly ugly and difficult to configure for the actual geolocation. I have to clean it up first.