GregDMeyer/IT8951 #11: add tkinter to requirements

mankyd, 2020-07-08:
display.py imports tkinter:

```
import tkinter as tk
```

It should be added to requirements.txt

mankyd, 2020-07-08:
Actually, after some more investigation, the solution for getting it working my virtual-env was simply:

```
sudo apt install python3-tk
```

GregDMeyer, 2020-07-13:
yeah, tkinter should be included in most python distributions by default, and by my understanding, if it's not, it can't be solved by pip (and thus requirements.txt).

one thing I can do is to move the tk import closer to where it is used, so it isn't imported unnecessarily. that way, most users of the library won't need it, unless they use the specific virtual display functionality.

mankyd, 2020-07-13:
Conditional import would be great. Maybe add a line to the readme stating that tkinter should be setup.

Jacob-Bishop, 2020-09-06:
Seconding that a line in the readme would be nice. 
On a fresh raspberry OS install (which I'm guessing is a common use case for this module?),  `libopen2-7`, `'libtiff-5`, and `python3-dev` also needed to be installed via apt.

FutureSharks, 2020-09-29:
My preference would be to remove the dependency on `tkinter` and have a separate `requirements.txt` for running the tests. I don't really want to install `python3-tk` and its X11 dependencies when they are only used for the tests 🙂

GregDMeyer, 2020-10-03:
OK, with the changes I just merged you should not need tkinter any more unless you are running the tests. Let me know if you find any issue with the change!

closed 2020-10-03 by GregDMeyer

Ken-mbira, 2021-10-10:
If it was added to the requirements file then maybe you could paste how it looks for us to just paste ourselves.