GregDMeyer/IT8951 #18: Is the 6 inch HD display supported?

FutureSharks, 2020-09-26:
This one: https://www.waveshare.com/wiki/6inch_HD_e-Paper_HAT

SirDagen, 2020-09-27:
Yes, it works. I had had it, but didn't like the reflective surface, returned it and went for the 9.7" https://www.waveshare.com/wiki/9.7inch_e-Paper_HAT

FutureSharks, 2020-09-27:
I tried it but get the error `communication with device failed`:

```
# python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from IT8951 import constants
>>> from IT8951.display import AutoEPDDisplay
>>> display = AutoEPDDisplay(vcom=-2.48)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/IT8951/display.py", line 219, in __init__
    epd = EPD(vcom=vcom, bus=bus, device=device, data_hz=spi_hz)
  File "/usr/local/lib/python3.7/dist-packages/IT8951/interface.py", line 33, in __init__
    self.update_system_info()
  File "/usr/local/lib/python3.7/dist-packages/IT8951/interface.py", line 103, in update_system_info
    raise RuntimeError("communication with device failed")
RuntimeError: communication with device failed
>>>
```

Any idea?

GregDMeyer, 2020-09-28:
Hmm. Does the example code from Waveshare (e.g. [here](https://github.com/waveshare/IT8951/)) run OK? If it doesn't, it's probably a hardware thing (did you set the switches on the board to use SPI for communication?).

If that code does work, then maybe try adjusting the SPI frequency (see the README). Perhaps the default value is too high for your device.

FutureSharks, 2020-09-28:
I got it working but not clear how. Is it possible to leave the display in a non-working state from using `ctrl-c` at some point?  I was also testing with another display driver and papertty also.

GregDMeyer, 2020-09-29:
huh. yeah, I have found in the past that sometimes using it with another display driver can cause problems. in particular, if you use code that communicates with the device via the `bcm2835` C library instead of using the kernel for SPI, it seems that the kernel relinquishes control of SPI. Since this library relies on kernel SPI, it will not work after that. So as you found, a reboot should get it working again :)

closed 2020-09-29 by GregDMeyer

FutureSharks, 2020-09-29:
> in particular, if you use code that communicates with the device via the bcm2835 C library instead of using the kernel for SPI

OK, I think that might have been it. Thanks for the reply.

weow147, 2021-12-26:
I keep getting this error as well.  if i turn off/on SPI it seems to recover.
```
sudo dtparam spi=off
sudo dtparam spi=on
```