GregDMeyer/IT8951 #24: No such file or directory: '/dev/spidev0.0' when trying to run the example

matyasf, 2020-10-13:
When I try to run the example (virtual display works fine) I get the following error

```
Initializing EPD...
Traceback (most recent call last):
  File "test/integration/test.py", line 56, in <module>
    main()
  File "test/integration/test.py", line 32, in main
    display = AutoEPDDisplay(vcom=-2.06, rotate=args.rotate, spi_hz=24000000)
  File "/usr/local/lib/python3.7/dist-packages/IT8951-0.1.0-py3.7-linux-armv7l.egg/IT8951/display.py", line 219, in __init__
    epd = EPD(vcom=vcom, bus=bus, device=device, data_hz=spi_hz)
  File "/usr/local/lib/python3.7/dist-packages/IT8951-0.1.0-py3.7-linux-armv7l.egg/IT8951/interface.py", line 26, in __init__
    self.spi = SPI(**spi_kwargs)
  File "IT8951/spi.pyx", line 51, in IT8951.spi.SPI.__cinit__
    self.fd = os.open(fd_path, os.O_RDWR)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/spidev0.0'
```

matyasf, 2020-10-13:
the solution: Enable spi with `raspi-config`

closed 2020-10-13 by matyasf

GregDMeyer, 2020-10-14:
Thanks for opening the issue---I'll include that in the README for other folks!