GregDMeyer/IT8951 #25: Mirror image inversion

dankarlin, 2020-10-20:
Writing an image to the screen, the image is left/right transposed. Rotation and flipping just re-orient the mirror image.

GregDMeyer, 2021-01-04:
Sorry for the delay in responding to this---unfortunately I can't reproduce the bug; on my device the image is displayed correctly when I call that function. Have you figured out the problem since you opened the issue? If you have more details I'm happy to reopen the issue and try to get to the bottom of it!

closed 2021-01-04 by GregDMeyer

Mari0nV, 2021-06-13:
I got the same issue, I don't know why.
I fixed it by replacing the function *_get_frame_buf* in *display.py* by:
```
def _get_frame_buf(self):
        return self.frame_buf.transpose(Image.FLIP_LEFT_RIGHT)
```

GregDMeyer, 2021-06-14:
That is so strange. Let me know if you find out any further why this is happening, it's so weird that it only happens for a few people!