GregDMeyer/IT8951 #27: Using Driver for a 7.8in waveshare EPD

grooverkm, 2020-11-18:
Sorry up front for beginner questions, I am trying to use the driver with a 7.8in EPD for a Very Slow Movie Player.  RaspPi 4 2GB.  I believe I have the driver built andinstalled correctly (I can run the test script ok) and I have added it to the correct import in the SlowMovie script.  It runs until the first command init.Clear then fails saying command not found.  If I comment the init command out, it just fails at the next one.  Is there something I am missing or haven't installed correctly?  Thanks in advance and apologies if this is the wrong place for such a question.

mweschan, 2020-12-27:
@grooverkm I'm actually starting on the exact same project with the 7.8in waveshare. Were you able to get this resolved?

grooverkm, 2020-12-27:
@mweschan unfortunately not.  Eventually went with the proven 7in5_V2.  Watching various projects though as I still have the first screen waiting a fix.

mweschan, 2020-12-28:
I managed to get it working! It took a few adjustments, but the SlowMovie script is working on my 7.8in screen. When I have a free moment I'll do a little write up to explain the script changes

GregDMeyer, 2021-01-04:
Thanks for the issue! @mweschan , it will be great if you can post anything relevant to this issue when you have it written up, in case other people want to come looking :) Thanks!

closed 2021-01-04 by GregDMeyer

HankMcB, 2021-03-22:
> I managed to get it working! It took a few adjustments, but the SlowMovie script is working on my 7.8in screen. When I have a free moment I'll do a little write up to explain the script changes

@mweschan Would be helpful to me as well.

mweschan, 2021-03-25:
Hi @HankMcB. Sorry that I haven't gotten around to doing a full write up yet, but here's the 2 minute overview. The first step is to install this IT8951 program and make sure it works. If you can get this installed and are able to successfully run the test file (IT8951/test/integration/test.py) and display the image on the epaper screen then the rest will be fairly straightforward. 

Once I got the test file working and displaying the test image on the e-paper screen, I modified it to point to the screengrab that the slow movie player program was creating rather that then test image. Lastly, I modified the slow movie player program so that after it created the image, it just called the test script to display the image on the screen.

Let me know if you get stuck and I can try to help out