GregDMeyer/IT8951 #28: Partial refresh images

jokober, 2020-11-18:
Since waveshares IT8951 driver doesn't support partial refresh I was hoping this implementation might be able to do partial refreshes of images. Unfortunately I couldn't find any example code in the tests though. 

Is it possible to partially refresh images?

GregDMeyer, 2020-11-18:
Yeah you can, for sure! the partial refresh doesn't care what it is displaying, I just used text as an example. If you display an image using the AutoEPDDisplay class (like in the example), then modify the image in `display.frame_buf` and then call `partial_update`, it should update only the parts that have changed.

closed 2020-11-18 by GregDMeyer

jokober, 2020-11-18:
It worked like a charm. Thanks for your great work!

GregDMeyer, 2020-11-18:
awesome, glad to hear it! :)

GregDMeyer, 2020-11-18:
btw the partial update usually updates a rectangle around the area that was changed, but if you use the "DU" update waveform it will only update the exact pixels that were changed!