GregDMeyer/IT8951 #29: Contrast drifting to black

t5b6de, 2020-12-12:
Contrast is drifting to darker image, after few full refreshes (doesnt matter if i use draw partial or draw full) contrast drifts to dark.

After while the whole display itself seems to crash. i have to reset the board otherwise the panel itself will be destroyed!

I cannot reproduce that behavior with the test program from ITE (E-LINK-TCON-DEMO) but with your implementation it does occur after several "Full"-Updates. First sign is, display gets darker. Small partial Updates are not affected.
See attached image. Clocktime is upated partial since background is updated full.

Even Display Mode INIT leaves a grayish Screen. I have to reset whole IT8951 Driver Hat.

I have rewritten that Code (not fully complete) into C# - Same issue

After every "full" or big partial Update, display gets darker and darker. Small Parial Updates such as Time OSD are stay correct:
[image]

kind regards

GregDMeyer, 2020-12-14:
oh no! I've never experienced this issue so I'm not sure how to help debug it. Did you try adjusting the VCOM value in the script? It is different for each display. I think the Waveshare code has a different VCOM value hardcoded in. My display came with a "best" value on a tag on the display cable, it might be worth checking if yours has that and setting it.

t5b6de, 2020-12-14:
Hi,
thanks for your response.

vcom is definetly correct. after refresh i read it back and got my -1.46 Volts.
Hard coded default value on the it8951 firmware is -2.6 Volts after reset.

one panel is already destroyed, but i got a free replacement. i contacted the seller for support, because i didn't get any response from waveshare.

normally it should not possible to destroy your panel with that software, even if you do manipulate the vcom voltage.
the datasheet for the it8951 which is public available is not complete unfortunately. registers are not explained and so on.

There is another register for setting extended driving capabilities of the it8951, but that even accellerated the process. after 2 or 3 refreshes the display did not respond (the hat does response fine), so on softwareside there's no way to detect a crashed panel.

That courios thing is, a reset from gpio pin, or button on that thing does resolv that problem.
But then i have to do the full re-init process.

Did you ever try to make a clock with that thing and update only with partial refresh (and full update every 10 partial refreshes) and re-init all 100 partial refreshes? (that was my plan for that photo frame.
how does your display respond?

kind regards
Thomas

GregDMeyer, 2020-12-14:
Hmm, yeah, during the holidays last year I made an animated Christmas tree on it, and had it doing partial refresh every few seconds for at least a day, with no problems. Must have been tens of thousands of partial refreshes in a row, and it worked fine.

I agree, destroying your panel is definitely not something that should be possible!! Have you tried with maybe a different Pi? Perhaps it's something in the hardware? I agree, the datasheet unfortunately is not very informative :/

Obviously I would be very interested in figuring this out so nobody else runs into this problem! Let me know if you figure anything further out, or if there's anything you want me to test on my device!

t5b6de, 2020-12-14:
I'm currently using a Raspberry pi zero wh. 

The first display did work for about one by two weeks. this one shows the first issues about 2 days after first use.. much faster.

This is the wiring, pretty simple. Three amps step down voltage regulator, adjusted to 5V, (measured 4.97V)
connected to the supply pins on the back side of the PCB. So the Rpi is supplied through the GPIO-Pins.

[image]

i've ordered a brand new raspberry pi 4 with 2GB RAM. I will test it with that one.

But i do not have much hope that this will solve the Problem. Positive sideffect: the rpi4 has MUCH MORE processing power...

GregDMeyer, 2020-12-14:
OK! For now, I'm not sure what to say. Let me know how things turn out with the Pi4 (and have fun!). And certainly let me know if you figure anything else out.

t5b6de, 2020-12-15:
i hope it's okay if i post something new infos here step by step....

i couldn't wait any longer cause my reseller said (without any further comment) i should send it back and they will investigate that issue.

so i decided to remove my rpi4 from my 3d printer server and use it temporarily for my project.

it ran about 90 minutes without ANY issue. so that is much more than with the rpi zero. But i've connected it with the delivered cable, not as a hat.
But that last display update was very creepy. it startet normally but endet up extremly slow.

So a mostly blank image was left. only black pixel remained black.
Thats a new fault. never seen that before.

after a press on the reset button on the pcb, it worked again fine.
I've the bad feeling that the whole delivery from the supplier has a fault or something. 

so no further idea what that could be.

kind regards.

SirDagen, 2020-12-15:
Hi, something similar happened to me. The display became darker and darker, the contrast lower and lower. I then disconnected the power supply and restarted and it was fine again.

However I then also changed the following in my programme code. Now I do a complete clear every night at 4:00 am:

```
                        # black screen
                        blscr = Image.new('RGB', (epd.width, epd.height)).convert("RGBA")
                        display.frame_buf.paste(blscr) # .rotate(90, expand=True)
                        display.draw_full(constants.DisplayModes.GC16)
                        time.sleep(1)
                        # white screen
                        display.clear()
                        time.sleep(1)
```

And I also always use sleep mode now.

```
                        epd.sleep()
                        time.sleep(60)
                        epd.run() # run again, after sleep
```

Since then I have had no more problems. I don't know, maybe it's the sleep mode.

P.S. Last year I destroyed a display without a controller because I didn't use the sleep mode. In the end the display only added black pixels, but couldn't clear any more....

t5b6de, 2020-12-16:
Hey,

i've added a sleep command after every interaction with the display controller. So after every Display Update a sleep() is called.

I've rewritten this entire thing in c# for mono on raspberry. because python is definetly not my favourite language. im writing mostly in c#. but dotnet core is not running on the raspi zero. so i had to use mono, performance is much worse compared to python or dotnet core.

so far so gut, i had no further problems after about half an hour. No further darkening, etc. I hope that is the solution.

t5b6de, 2020-12-18:
The sleep command is the final solution. The first panel was indeed defect. This only drifted to darker and bad contrast.

now thats my final Setup:
[image]
[image]

Kind regards, and Thanks for that awesome code here!

closed 2020-12-18 by t5b6de

GregDMeyer, 2020-12-18:
awesome, glad to hear it! I think I will add an automatic sleep() call after the updates in the AutoEPDDisplay class, so hopefully fewer people run into this issue.

SirDagen, 2020-12-22:
Hi Greg, nice to hear from you. If auto sleep comes with some disadvantages (e.g. process time) could you consider making the use of auto sleep optional? I sometimes do a quick slideshow, which sleep might interfere with. 

Wish you and all the fellow epaper coders a merry christmas.

GregDMeyer, 2020-12-22:
Yes, for sure!! I will make it optional :) But probably set auto sleep as the default, so if people write some quick code without paying too much attention, they don't potentially break their display.

Thank you for the feedback, and enjoy the holidays!!

gadaga, 2020-12-24:
@t5b6de can you provide STL for you frame? I love it

gadaga, 2020-12-24:
> The sleep command is the final solution. The first panel was indeed defect. This only drifted to darker and bad contrast.
> 
> now thats my final Setup:
> [image]
> [image]
> 
> Kind regards, and Thanks for that awesome code here!

@GregDMeyer can you provide STL for you frame? I love it

t5b6de, 2020-12-24:
Will upload that on thingieverse soon. assembly is not very optimal but working.

You need a 13x18cm glass sheet, then there is a front spacer between panel and glass sheet. then the panel, then the inner frame. where all the other components are screwed on. you have to drill the holes on the Rasoi to 3mm because I'm using M3 screws.

But I will add a small assembly information on that page, but first:

have nice Christmas Days, stay healthy!

t5b6de, 2020-12-26:
That's my Xmas-present for you
https://www.thingiverse.com/thing:4697945

Yuv3n, 2021-12-07:
@GregDMeyer Have you added the autosleep function yet? I havent found it in the source

GregDMeyer, 2022-01-12:
Hi,

Sadly, my epaper display physically broke, and I can't test any code changes until I get a new one. I may just buy a cheap one soon so I can test again. Sorry about that! I'll update when I am back in action.

txoof, 2022-01-12:
@GregDMeyer If you're open to donations/patreon/paypal/whatever I'd happily kick in some coin for replacing your screen. I really appreciate your code and all the work you've put into it. 

Without your code, my [PaperPi](https://github.com/txoof/epd_display#readme) project would be dependent on the WaveShare Demo code for it8951 boards and that would be **AWFUL**. 

I've got three screens running around the house and office that use your code all day, everyday.

GregDMeyer, 2022-02-14:
Thanks so much, that's really sweet :-) It really means a lot and motivates me to maintain the code!

I plan to order a new display soon (and maybe I'll add a donation link, we will see, thanks again!). Just wanted to check: does anyone know if there's a good way to flash new firmware onto the IT8951? I would like to just buy a new raw display and not have to pay the extra ~$60 for a new IT8951 HAT as well, but it seems (from what I've heard) that I can't plug a display with a different resolution into my existing HAT without modifying the firmware to match the new resolution. 

(I would just buy the identical display again, but it seems waveshare no longer carries the 6 inch 800x600 display).

Let me know if anyone has any thoughts, and hopefully then I can order it soon! Thanks everyone.

txoof, 2022-02-14:
It may be possible to flash new firmware onto the control boards, but according to the WaveShare support, this isn't possible.

I recently toasted a board for the 9" screen and ordered a loose board direct from WaveShare. They delivered a board for the next size up (12"?) and it wouldn't work with my 9" display.

After about 100 emails, they shipped a new board with the 9.7" firmware. 

I have successfully dumped the 9.7" firmware and I've been meaning to try to flash it to the 12" board, but have t gotten around to it yet. I'm also a little worried that if I do something wrong I could damage the display. I don't know enough about what I'm doing.

This sounds like a perfect opportunity to let some smoke out ...

I'll look for a donation link if you set one up.