GregDMeyer/IT8951 #31: Waveshare 6inch HD

pierrolivier, 2021-01-13:
Hi,
I just received my new Waveshare 6 inch HD display with a resolution of 1448 × 1072px. Unfortunately, I cannot use it with your package. The screen is not updating correctly. Only the left part is updated and the image is visible twice.
Have you got any idea about this?
Thanks by advance

[image]

GregDMeyer, 2021-01-13:
Oh no! Well, this certainly appears to be a bug. I'm not exactly sure how to help debug, since I don't have the 6in HD device. But we can try.

One first question---is the test code distributed by Waveshare itself functioning correctly? It looks like on [the wiki](https://www.waveshare.com/wiki/6inch_HD_e-Paper_HAT) they advertise [this repository](https://github.com/waveshare/IT8951-ePaper) for example C code. Checking that that code works can help rule out any hardware problem. (Note that you might need to restart your device after running their code, if you want to use this library again).

Also, just to check, you are using the HAT?

Thanks for the bug report! We will try as best as possible to figure out what's going on :-)

pierrolivier, 2021-01-13:
I've just test the display with their test code and I've got the same problem.
I use the IT8951 Driver HAT from my old 6 inch (800*600). Maybe the problem come from here. Their website site ins't very clear about this. I've seen there is a new HAT (B).
When I start the test code, it say that the panel has a definition of 800x600.

Thanks for your help!
[image]

GregDMeyer, 2021-01-13:
Ah, yes, that is certainly why you are running into this. The HAT thinks that it is still driving the 800x600 display. I think it may be possible to re-flash your HAT's firmware to be compatible with the new display, but I don't know how to do that. In any case, it isn't an issue with this library, so I will close the issue.

Good luck!

closed 2021-01-13 by GregDMeyer

aaronvizzini, 2022-04-28:
@pierrolivier Did you happen to either get the old HAT working or get a new HAT for the 6inch HD display? I wanted to know if this library would work correctly with the HD version of the Waveshare display, before I went ahead and ordered one. Thanks.

GregDMeyer, 2022-05-03:
FYI, I just (finally) replaced my old 600x800 display with the 6 in HD version, and it seems to be working great with the library.

aaronvizzini, 2022-05-04:
Good to know. I acquired one as well, I only got around to briefly testing it with Waveshare's Windows software, just to make sure all the hardware was working. Thanks for putting this together.