GregDMeyer/IT8951 #32: Weird black stripes across display

doebi, 2021-01-19:
Hello everyone!
First of all thanks GregDMeyer for this project.

I recently received a Waveshare 7.8 HD with a resolution of 1872x1404 and managed to get it working with waveshare's sample code and this library likewise.

But one weird problem is consistent among both approaches: I have perfectly regular black stripes on my display. Since they are consistent regardless which library I use I assume it is not a software related bug. Nevertheless I am asking here because I do not know of a better place. There seem to be a lot of epaper experts around.

[image]

Here is what I found so far:
- the stripes have not been there when the display was unpacked but appeared on first successful write
- they remain when "clearing screen"
- I was guessing on faulty wiring and skipped the extension cord -> no change
- variation in VCOM value also have no effect

Has anybody experienced something similar or has an explanation for this? Any help is appreciated. Thanks.
If it turns out to be faulty hardware I will try to get it replaced. :/

GregDMeyer, 2021-01-19:
Ugh, that's frustrating. I have not personally experienced such a problem, but I do think it is almost certainly something faulty in the hardware. I would ask Waveshare for a replacement. :-(

doebi, 2021-01-19:
Thanks for your answer, although it is not related to your software.

Yeah, frustrating indeed. I got it from a retailer in Germany and I will contact them promptly. :/

Thanks again for your opinion.

closed 2021-01-19 by doebi