GregDMeyer/IT8951 #33: /dev/spidev

Manudulux, 2021-01-22:
FileNotFoundError : Error 2 No such file or directory: '/dev/spidev0.0 '

It looks like this required file is no longer available from the updsted Kernels anymore.
Any idea ho to solve this?

PS: I am the newest newbie amongst the newbie. Just hope my question makes some sense.

GregDMeyer, 2021-01-22:
Thank you for opening the issue! I think this may mean that you don't have SPI enabled on your device. Try running `raspi-config` (assuming you are on Raspberry Pi) and under "Interface" enable SPI. Let me know how it goes! (Perhaps I should make this more clear in the `README`).

Manudulux, 2021-01-23:
Hi Greg, first of all thanks so much for your prompt answer. This is very much appreciated. You know this feeling of getti g help when you are stuck. Second your README instructions are crystal clear   No need to amend them. Third the issue with spidev0.0 is new. I found several forums where this is mentionned, all starting in November 2020. I am pretty sure that something has changed in the kernel. Will keep investigating and share if I come to some conclusion. Thanks again for your work and your support.

Manudulux, 2021-01-24:
Hi Greg, I finally found the solution under https://www.raspberrypi.org/forums/viewtopic.php?t=136912
I do not understand the background fully but a DT_param line in the config,txt was the cause of the issue.
And the false entry was probably creating by myself when I tried codes designed for other epaper screen.
This was a user interface problem. 
Sorry for bothering you with this and thanks again for your work and support.
I leave this entry for te next newbie coming along and unavoidably doing a similar issue due to the lack of documentation of waveshare on this particular screen.
Happy coding to all.

GregDMeyer, 2021-01-26:
No problem, thanks so much for posting the solution so others can see it! Enjoy :-)

closed 2021-01-26 by GregDMeyer