GregDMeyer/IT8951 #35: Test fail with 9.7" Panel

txoof, 2021-06-22:
I would like to try and use this library with the 9.7" panel, but the tests keep erroring out with `RuntimeError: communication with device failed`.

 I have a pipenv virtual environment with the proper requirements installed.
```
ipykernel==5.5.5
 <<removed ipykernel related packages>>
IT8951==0.1.0
Pillow==8.2.0
RPi.GPIO==0.7.0
```
The test code provided by waveshare appears to work as expected; it displays a graduated gray field, a line drawing and then an image. The output of the waveshare IT8951 executable outputs the following while displaying images:
```
$ sudo ./IT8951 0 0 17.bmp
****** IT8951 ******
Panel(W,H) = (1200,825)
Image Buffer Address = 11A1E0
FW Version = SWv_0.2.1T
LUT Version = M841
VCOM = -1.50V
IT8951_GUI_Example

*****************************************
BMP_cfSize:55864
BMP_cfoffBits:62
BMP_ciSize:40
BMP_ciWidth:800
BMP_ciHeight:558
BMP_ciPlanes:1
BMP_ciBitCount:1
BMP_ciCompress:0
BMP_ciSizeImage:d9fa
BMP_ciXPelsPerMeter:2e25
BMP_ciYPelsPerMeter:2e25
BMP_ciClrUsed:0
BMP_ciClrImportant:0
*****************************************

total_length:55802,55802
bytesPerLine = 100
imageSize = 55800
skip = 0
```
I've adjusted the test.py as suggested and changed the VCOM value to `-1.50`.

When I run the `test.py` in the pipenv virtual environment, I get the following output:

```
$ python3 test.py                                            
Initializing EPD...
Traceback (most recent call last):
  File "test.py", line 56, in <module>
    main()
  File "test.py", line 32, in main
    display = AutoEPDDisplay(vcom=-1.50, rotate=args.rotate, spi_hz=24000000)
  File "/home/pi/.local/share/virtualenvs/IT8951-cl9zA0KD/lib/python3.7/site-packages/IT8951/display.py", line 219, in __init__
    epd = EPD(vcom=vcom, bus=bus, device=device, data_hz=spi_hz)
  File "/home/pi/.local/share/virtualenvs/IT8951-cl9zA0KD/lib/python3.7/site-packages/IT8951/interface.py", line 33, in __init__
    self.update_system_info()
  File "/home/pi/.local/share/virtualenvs/IT8951-cl9zA0KD/lib/python3.7/site-packages/IT8951/interface.py", line 103, in update_system_info
    raise RuntimeError("communication with device failed")
RuntimeError: communication with device failed
```
It appears that no SPI data is being returned when the board is queried. With constants.Commands `GET_DEV_INFO`. 

A check of the `constants.Commands` against the WaveShare `IT8951.h` shows a slight difference in the addresses used when calling various commands. I've tried updating `constants.Commands` to match the WaveShare Values, but that does not remedy the problem.

The WaveShare `IT8951.h` version is below:
```
//Built in I80 Command Code
#define IT8951_TCON_SYS_RUN      0x0001
#define IT8951_TCON_STANDBY      0x0002
#define IT8951_TCON_SLEEP        0x0003
#define IT8951_TCON_REG_RD       0x0010
#define IT8951_TCON_REG_WR       0x0011
#define IT8951_TCON_MEM_BST_RD_T 0x0012
#define IT8951_TCON_MEM_BST_RD_S 0x0013
#define IT8951_TCON_MEM_BST_WR   0x0014
#define IT8951_TCON_MEM_BST_END  0x0015
#define IT8951_TCON_LD_IMG       0x0020
#define IT8951_TCON_LD_IMG_AREA  0x0021
#define IT8951_TCON_LD_IMG_END   0x0022

//I80 User defined command code
#define USDEF_I80_CMD_DPY_AREA     0x0034
#define USDEF_I80_CMD_GET_DEV_INFO 0x0302
#define USDEF_I80_CMD_DPY_BUF_AREA 0x0037
#define USDEF_I80_CMD_VCOM                 0x0039
```

Beyond this, I'm getting out of my depth and blindly stumbling around. Do you have any suggestions as to how to diagnose and remedy this problem?

input-variable, 2021-06-22:
I've come across this issue as well. It only ever occurs when there has been an error in my code and it had to halt before closing communications. I usually just restart my Pi and it seems to fix it.

txoof, 2021-06-23:
@input-variable 
Thanks! A reboot resolved this problem. 

It appears that the SPI bus gets disturbed sometimes (*looking at you WaveShare demo code*) and read/write is busted. Any ideas what causes this or how to fix it without a reboot?

input-variable, 2021-06-23:
I'm really not too sure what's going on in the background with all of this, I guess it may be possible to reset the SPI interface instead of the entire device. I actually haven't had this error in a while since I stopped using the WaveShare demo code. Maybe you could try that?

txoof, 2021-06-28:
resolved with reboot

closed 2021-06-28 by txoof