GregDMeyer/IT8951 #36: IT8951 Module hangs forever on init when no screen is connected

txoof, 2021-07-05:
How to reproduce: Remove IT8951 driver board from Pi, call: `myepd = AutoEPDDisplay(vcom=-1.93)`

The `AutoEPDDdisplay` object will wait apparently forever for the spi read/write to complete.

```
IT8951/interface.py in update_system_info(self)
     97         Get information about the system, and store it in class attributes
     98         '''
---> 99         self.spi.write_cmd(Commands.GET_DEV_INFO)
    100         data = self.spi.read_data(20)
    101 
```

GregDMeyer, 2021-07-06:
I am happy to look into fixing this, but just had one quick question---is there a use case for running the code without a board connected?

txoof, 2021-07-07:
Thanks for looking into this.

I'm using this as part of a user facing [project](https://github.com/txoof/epd_display). The target audience is not necessarily technical. I'm adding support for IT8951 boards.

 If there is a way to detect when there is no screen detected, that information can be used in informative error messages to assist in troubleshooting.

Ideally the auto EPD method would throw an exception if the SPI write process fails.

grob6000, 2021-07-10:
Here's the problem: suggest we throw an exception on wait_ready if it takes more than 5 sec or so. Applications can pick it up.
In spi.pyx:

    def wait_ready(self):
        '''
        Wait for the device's ready pin to be set
        '''
        while not GPIO.input(Pins.HRDY):
            sleep(0.001)

    def transfer(self, int size, int speed):
        '''
        Perform an SPI transaction of *size* bytes on the preallocated read and write buffers.
        '''
        cdef spi_ioc_transfer tr

        self.wait_ready()

txoof, 2021-07-11:
I'm not super familiar with the `[signal](https://stackoverflow.com/questions/492519/timeout-on-a-function-call)` module, but it looks like might be an appropriate solution.

```
def handler(signum, frame):
    # probably need a better Exception here
    raise Exception("Timeout")

def wait_ready(self, wait=10):
    '''
    Wait for the device's ready pin to be set
    '''
    signal.signal(signal.SIGALRM, handler)
    signal.alarm(wait)
    try:
        while not GPIO.input(Pins.HRDY):
            sleep(0.001)
    except Exception as e:
        # again, probably a more appropriate exception
        raise Exception(f'Timeout: device ready pin was not set after {wait} seconds')
```

closed 2022-02-20 by txoof

GregDMeyer, 2022-05-04:
fixed by a8df9840394b627f5f05870340ab88476fda4a48