GregDMeyer/IT8951 #40: Text displays backwards on eInk display only

ScienceAndCats, 2021-09-24:
Using the virtual display shows text correctly, but when using the test.py program, any text displayed on the eink display is backwards. Images seem fine.

GregDMeyer, 2021-09-30:
This is a very weird bug. Seems to be the same as #25 and #37 , but I can't reproduce any of these issues on my device.

See those issues for a workaround; other than that, all I can say is that if you figure out what is going on, please let me know!!

ScienceAndCats, 2021-10-02:
Thanks for getting back to me! Will do!

GregDMeyer, 2022-05-02:
This is "fixed" (in that you can set mirror=True to account for it) by 2a81167a36c5c3673cb9a73393b364682b0e644a. I think it seems that it is just an issue with certain particular boards from waveshare, that they flip all their output for some reason.

closed 2022-05-02 by GregDMeyer

txoof, 2022-05-03:
Thanks! We'll test this out.