GregDMeyer/IT8951 #41: Command "python setup.py egg_info" failed with error code 1

sperez91, 2021-10-01:
I've been getting the following error when executing **pip install ./**:
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-WsATgY/
See screenshot.

I've been googling fixes for this but nothing works. Anybody can help?

[image]

GregDMeyer, 2021-10-04:
Hi there! There is an easy fix, try running with `pip3` instead of `pip`, this should target Python 3 instead of Python 2. Sorry about that, I will update the README to make this more clear.

sperez91, 2021-10-04:
Thank you so much! It works (:

closed 2021-10-04 by sperez91