GregDMeyer/IT8951 #42: How to manually set resolution when not detected properly when creating an AutoEPDDisplay object?

txoof, 2021-10-04:
I had to buy a loose driver board from WaveShare because I toasted the one supplied with a 9.7" panel. The board they sent works fine, but I think it's set for 1600x1200 rather than the 1200x825 resolution of the 9.7" panel.

Up until I killed my stock driver board, I used the following code and it worked properly:
```
from IT8951.display import AutoEPDDisplay
from IT8951 import constants as constants_HD
myepd = AutoEPDDisplay(vcom=self.vcom)
```
Unfortunately, allowing AutoEPDDisplay to pull the width and height from the driver board results in AutoEPPDisplay pulling the resolution from the board and setting it to 1600x1200. All the images are clearly not formatted correctly; "partial...update" test falls mostly off the screen and the image of Tux is only about 1/4 on the display. Formerly this worked just fine.
```
$ python3 ./test.py                                                 
Initializing EPD...
VCOM set to -1.9
System info:
  display size: 1600x1200
  img buffer address: 119F00
  firmware version: WS_v.0.1
  LUT version: 6M14T
```

I can see that the 
`__init__` in `AutoEPDDisplay()` class sets the property `epd=EPD()` and `AutoDisplay()` gets passed the `epd.width, epd.height` values. What I can't figure out how to do is properly pass in the width and height arguments so that `EPD()` gets them.

I tried to set the width and height manually, but I'm clearly not doing things correctly as the resolution reports that it is correct, but the images are still not scaled correctly:
```
display =  AutoEPDDisplay(vcom=-1.90, rotate=args.rotate, spi_hz=24000000)
display.epd.width=1200
display.epd.height=825
```

```
$ python3 ./test.py
Initializing EPD...
VCOM set to -1.9
System info:
  display size: 1200x825
  img buffer address: 119F00
  firmware version: WS_v.0.1
  LUT version: 6M14T
```

Is there a way to do this?

Alternatively, do you have any idea how to bang the resolution into the board firmware?

txoof, 2021-10-05:
@GregDMeyer 
I'm diving out into the weeds trying to solve this problem. Can you confirm that the width/height is stored at this memory address: [`LUT0EWHR  = DBASE + 0x00`](https://github.com/GregDMeyer/IT8951/blob/907e587d48e4de18f1d9bfc248e66bc352ee478a/IT8951/constants.py#L78)?

I think I might be able to use the E-LINK-TCON Demo app to write the appropriate value to the firmware, but I'd rather not hoze over another board if I can help it.

Thanks again for any advice you can offer.

GregDMeyer, 2021-10-05:
Hi @txoof ! Thanks for opening the issue!

I don't have any experience writing the firmware to the IT8951, unfortunately, so I don't think I can help you there. I think, as it seems you suspect, the issue is that even though you are setting the height and width correctly in the Python class, the IT8951 is still *expecting* data with the wrong dimensions, so it will still display incorrectly :-(

Did you try emailing waveshare support to ask about how to put new firmware, by the way?

txoof, 2021-10-05:
@GregDMeyer Thanks for getting back to me. 
I did email WaveShare, but they're out on holiday for a week. 

It's good to know that the problem is with the board. That helps narrow down the problems. Fortunately I have three driver boards to play with, one that definitely works with a 9.7, one that appears to be set wrong, and the one I likely toasted. I'm hoping that I can read out the memory and find the differences between the working board and the one that's set to the wrong resolution. 

In your code, the constants pull WxH from a LUT. I know it's a stretch, but do you think I can read/write the correct data there, or is the LUT just a pointer to some other memory location?

GregDMeyer, 2021-10-05:
By the way, the line you mention originated from the .h file here, hopefully it can help!

https://github.com/waveshare/IT8951/blob/master/IT8951.h#L104

txoof, 2021-10-05:
that is super helpful.

GregDMeyer, 2021-10-05:
yeah, unfortunately I don't know exactly how that would work further than what is contained in the .h file there! Good luck, sorry I wasn't able to be more helpful!

txoof, 2021-10-05:
Maybe the folks over at the papertty project can give some insights. They're hip-deep into the hardware over there.

closed 2021-10-05 by txoof