GregDMeyer/IT8951 #46: Runs once with fresh install , breaks during test

j2r2jones, 2022-02-20:
Initializing EPD...
VCOM set to -2.06
System info:
  display size: 1872x1404
  img buffer address: 122520
  firmware version: WS_v.0.2
  LUT version: 8M14T_AF8221

Clearing display...
Displaying gradient...
Starting partial update...
  writing full...
  writing partial...
Displaying "images/sleeping_penguin.png"...
Traceback (most recent call last):
  File "/home/pi/IT8951/test/integration/test.py", line 56, in <module>
    main()
  File "/home/pi/IT8951/test/integration/test.py", line 50, in main
    t(display)
  File "/home/pi/IT8951/test/integration/test_functions.py", line 74, in display_image_8bpp
    img = Image.open(img_path)
  File "/home/pi/.local/lib/python3.9/site-packages/PIL/Image.py", line 2912, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: 'images/sleeping_penguin.png'

Any ideas here? The waveshare IT8951 drivers work fine. I'm just trying to get this display working with slowmovie

j2r2jones, 2022-02-20:
Here is the results from subsequent test run:
Initializing EPD...
Traceback (most recent call last):
  File "/home/pi/IT8951/test/integration/test.py", line 56, in <module>
    main()
  File "/home/pi/IT8951/test/integration/test.py", line 32, in main
    display = AutoEPDDisplay(vcom=-2.06, rotate=args.rotate, spi_hz=24000000)
  File "/home/pi/.local/lib/python3.9/site-packages/IT8951/display.py", line 219, in __init__
    epd = EPD(vcom=vcom, bus=bus, device=device, data_hz=spi_hz)
  File "/home/pi/.local/lib/python3.9/site-packages/IT8951/interface.py", line 33, in __init__
    self.update_system_info()
  File "/home/pi/.local/lib/python3.9/site-packages/IT8951/interface.py", line 103, in update_system_info
    raise RuntimeError("communication with device failed")
RuntimeError: communication with device failed

txoof, 2022-02-20:
@j2r2jones The first test errors out because it can't find `sleeping_penguin.png`. Do you ever see the gradient display and the "partial....update" display? 

The `test.py` script doesn't do any path checking and just assumes that `sleeping_penguin.png` is in a relative path. I can get the same results by running the following command:
`$ python3 ./test/integration/test.py`

Try move directly into `./test/integration/` and run the tests from there. The relative paths should work out better.

It's possible that when the test script crashed it left the SPI bus in a weird state. Try disabling SPI `sudo raspi-config` > Interface > SPI and then enabling it again. If that doesn't help, reboot and make sure you run the tests from the `test/integration/` directory.

I've also seen the `RuntimeError: communication with device failed` when the cable isn't seated properly or the board is toasted. I know you said the waveshare sample code works OK, but maybe double check that all the cables are seated properly and then test the WaveShare code again. If that works, there's more digging needed.

txoof, 2022-02-20:
I should also mention that switching between the WaveShare sample code and this code can have some extremely unpredictable results. The WaveShare code can leave things in a truly messed up state. You can try disabling the SPI bus and then enabling it again; that sometimes resolves this WaveShare issue. The most reliable solution is to reboot the pi and NEVER use the waveshare code again.

j2r2jones, 2022-02-20:
Thank you so much! Just turning the SPI off and back on seems to solve the issue. The test runs perfectly!

txoof, 2022-02-20:
@j2r2jones Glad I could help.

if this resolves your problem, you can save the maintainer the hassle of closing this ticket by choosing "close issue"

closed 2022-02-20 by j2r2jones