GregDMeyer/IT8951 #47: Add dependency requirements in setup.py file

robweber, 2022-02-22:
Would it be possible to add the dependency requirements in either the `setup.py` file so they can be pulled in when the library is built? This would help being able to include this library automatically as part of other projects without having to a) include this project reqs in the larger project or b) explaining they need to be installed separately. 

I understand there is a difference in install on RPI vs other systems for the purposes of testing. Would using platform filtering in this case solve that issue so it's still cross-platform? I can write up an actual PR but excluding the `x86_64` platform would filter out the Linux desktop systems but do the install on Arm architecture. 

```
setup(
....
install_requires = [
    'pillow',
    'RPI-GPIO; platform_machine != "x86_64"'
]
)
```

txoof, 2022-02-22:
This would be super helpful for my project as well. 

I'd be happy to test and validate.

GregDMeyer, 2023-11-04:
Fixed by 777d3473ec4ce64eee1326cf8e58dd17e30b8025

GregDMeyer, 2023-11-04:
(I was considering using platform filtering but I figured precisely getting the filtering right might be a headache, so I just added optional dependencies `rpi`. If this is annoying for some reason let me know, and we can try to do the best possible with platform filtering!)

closed 2023-11-04 by GregDMeyer