GregDMeyer/IT8951 #48: Rotate=flip + low_bpp + partial update is flawed.

mankyd, 2022-03-05:
I am trying to use partial update in DU mode to speed up interactions with my screen. My screen is also flipped, such that I initialize with `rotate='flip'`. I am finding that my screen updates improperly in this mode. In fact, I get the same behavior for any of the low-bpp modes.

To be clear, if I do full updates, it works. If I use non-low-bpp update modes, it works. If I don't set `rotate='flip'` it works.

What I am finding is that sometimes the screen updates, sometimes it doesn't. It seems to fail in particular if the updates extend all the way to what would be the top-left corner of the screen, what should be (0,0). Maybe the box_rounding function doesn't work when flipped?

I don't have great repro steps for this at the moment, unfortunately, but I will update if I can generate some.

mankyd, 2022-03-05:
I can confirm that changing `round_box` to `1` improves behavior overall (though other issues arise). Also, I can see through some logging that I've added that, in its default state, it tries to draw images larger than my target device, which causes it to misbehave.

```
2022-03-04 22:06:59,488 [hardware.einkdisplay] Screen Inited. Dimensions: (1872, 1404)
# printing out values from inside of `draw_partial`
buf:  (461, 10, 1852, 1318)  
diff_box:  (0, 0, 1872, 1408)
xy:  (0, 0)
dims:  (1872, 1408)
```

mankyd, 2022-03-05:
Last update: setting `round_box` to `4` (or simply commenting out the logic in partial_update) fixes the issue for me.

GregDMeyer, 2022-03-07:
Thanks for leaving this issue! My device broke but I have another one coming in the mail; I'll look into this once I have the new one.