GregDMeyer/IT8951 #49: Error when compiling the Cython code ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘use_tracing’;

aaronvizzini, 2022-04-28:
I'm encountering an error when running the build command:
`python setup.py build_ext --inplace`
```

IT8951/spi.c:21854:24: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘use_tracing’; did you mean ‘tracing’?
21854 |         return tstate->use_tracing && retval;
```

Full output here:
[error.txt](https://github.com/GregDMeyer/IT8951/files/8578537/error.txt)

Is this an error with something I'm doing, is there a workaround, or fix for this?

I was attempting to get this installed and running on Ubuntu (22.04 LTS) in order to test with a virtual display. 

I have Python 3.10.4 installed and Cython version 0.29.28

aaronvizzini, 2022-04-28:
After some more research it would seem @cetres's #45 resolves the compilation errors and allows me to run the test virtually on Ubuntu

GregDMeyer, 2022-05-02:
Fixed by 5f77bb853a262f110d6d1c75fab5a41b52de2882. Thank you for opening the issue!

closed 2022-05-02 by GregDMeyer