GregDMeyer/IT8951 #51: possibility of a micropython port?

ccoenen, 2022-05-08:
Hi! This project currently focuses on the Raspberry Pi in a few places. I was wondering, if this could also be changed to be more generic; with the ultimate goal being a Micropython implementation for the IT8951.

I'm currently trying to find out if micropython might be a good fit for the [m5paper](https://docs.m5stack.com/en/core/m5paper) which is an ESP32 + IT8951 in a pretty neat package.

GregDMeyer, 2022-05-10:
Hi there! I don't know a lot about micropython, but I think this could be possible. The main thing that relies on the RPi right now is the backend (the spi.pyx file). I think if this interface were rewritten for a different device, it could work! The other challenge is that the library currently relies on Pillow, and I'm not sure if that is supported. But that's only for the fancy "Display" class, if you're willing to work at a slightly lower level one could use the "Interface" class directly without a need for Pillow.

In summary I think it's certainly possible! Unfortunately I don't have the hardware or the time to work on this, but I am happy to provide guidance!

ccoenen, 2022-05-11:
I'm afraid my pyhthon knowledge ist rather limited. I'm currently starting to get into micropython to improve my skills. Feel free to close this issue :-)

GregDMeyer, 2022-05-11:
OK! Happy coding :-)

closed 2022-05-11 by GregDMeyer

RuofengX, 2022-06-18:
@ccoenen Hi coenen! Did you find a port of it8951 to micropython? I bought a M5 Paper device, and find a python library to drive this thing too.

ccoenen, 2022-06-18:
sadly, I did not. I am not very fluent in python, so I was pretty much out of luck. I ended up flashing M5Stack's UIFlow onto the device. It apparently is a somewhat old version of micropython internally. I'm using [Thonny](https://thonny.org/) to get my script onto it. Overall, I'm not terribly happy with the overhead, but in the end it still works fine, so I did not look much further.