GregDMeyer/IT8951 #53: GIF support

peterableda, 2022-09-01:
Do you have guidance on how to play gifs?

SungMatt, 2022-11-26:
I have this question too! Is it implemented using partial updating, or something else? In the [standard demo](https://github.com/waveshare/IT8951-ePaper) written in C, it appears to update without the flashes?

txoof, 2022-11-27:
You could extract each frame and then write them sequentially using a partial refresh.

[PIL and PILLOW](https://pythontic.com/image-processing/pillow/extract%20frames%20from%20animated%20gif) can help you with this.

You'll probably want to count the partial refreshes and then do a full refresh every N gif frames.

GregDMeyer, 2022-11-29:
Agreed---you should extract each frame of the GIF as a separate image and then use an appropriate update mode (I would guess DU4 is a your best bet if your particular screen supports it, since it allows some grayscale and is non-flashy. see [here](https://www.waveshare.net/w/upload/c/c4/E-paper-mode-declaration.pdf) for a description of the various modes). Overall it's a bit challenging to update to anything other than black and white without flashing, I think. But please let us know if you get it working nicely!