GregDMeyer/IT8951 #57: 7.8inch e-Paper HAT not working

Hoerli1337, 2023-10-10:
Hi!
I have the 7.8inch e-Paper HAT.
The display is connected to a Raspberry Pi 3B+ with freshly installed RaspbianOS.
The demo from the Waveshare homepage works and runs through without any problems.
Unfortunately nothing works with the Python script here :(

I installed everything according to instructions and apparently it worked without errors.
If I run the test scripts, unfortunately nothing works.
```
/home/pi/IT8951/test/unit# python3.9 test_img_manip.py
/home/pi/IT8951/test/unit# option parsing failed: Cannot open display:
[Here he is now stuck]
```

```
/home/pi/IT8951/test/integration# python3.9 test.py
Initializing EPD...
Traceback (most recent call last):
  File "/home/pi/IT8951/test/integration/test.py", line 58, in <module>
    main()
  File "/home/pi/IT8951/test/integration/test.py", line 34, in main
    display = AutoEPDDisplay(vcom=-2.15, rotate=args.rotate, mirror=args.mirror, spi_hz=24000000)
  File "/usr/local/lib/python3.9/dist-packages/IT8951/display.py", line 224, in __init__
    epd = EPD(vcom=vcom, bus=bus, device=device, data_hz=spi_hz)
  File "/usr/local/lib/python3.9/dist-packages/IT8951/interface.py", line 33, in __init__
    self.update_system_info()
  File "/usr/local/lib/python3.9/dist-packages/IT8951/interface.py", line 103, in update_system_info
    raise RuntimeError("communication with device failed")
RuntimeError: communication with device failed
```

I hope someone here has a solution to the problem.

Demo output:
```
/home/pi/Downloads/IT8951-ePaper/Raspberry# sudo ./epd -1.53 0
/***********************************/
bcm2835 init success !!!
/***********************************/
VCOM value:1530
Display mode:0
Panel(W,H) = (1872,1404)
Memory Address = 124850
FW Version = .v.0
LUT Version = M841
VCOM = -1.53V
A2 Mode:6
```

GregDMeyer, 2023-10-11:
Hmm, that's odd! Did you try rebooting the Pi? I have experienced some issues where running the waveshare demo first causes this code not to work. But if you reboot the pi and run this code without running the demo, it may work! Let me know if this helps.

txoof, 2023-10-12:
@Hoerli1337 +1 to GregDMeyer's comment. I've experienced a ton of problems with the waveshare demo taking over something in the GPIO bus and causing everything else to fail miserably. 

Try rebooting and then run the tests above.

Hoerli1337, 2023-10-12:
I somehow have absolutely no luck with this :(

To be on the safe side, I once again provided the Pi with a new image.
As luck would have it, Raspberry has now brought the new OS image live via the imager since yesterday.
Now Python 3.11 is preinstalled. Previously Python 3.9 was included.
So the driver can't be installed anymore.
python3-dev + python3-tk is installed.

Absolutely nothing else is installed on the Raspberry Pi.
Only the current system updates were installed and SPI was activated.

So that I can start the script at all, I have to send the following command:
`rm /usr/lib/python3.11/EXTERNALLY-MANAGED`

Here is what I have achieved as a root account:

```
# git clone https://github.com/GregDMeyer/IT8951.git
# cd IT8951/
# pip install -r requirements.txt
```
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from -r requirements.txt (line 1)) (0.7.1a4)
Requirement already satisfied: pillow in /usr/lib/python3/dist-packages (from -r requirements.txt (line 2)) (9.4.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```

```
root@raspberrypi3bplus:/home/pi/Downloads/IT8951# pip install ./
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/Downloads/IT8951
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: IT8951
  Building wheel for IT8951 (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [52 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-cpython-311
      creating build/lib.linux-armv7l-cpython-311/IT8951
      copying IT8951/constants.py -> build/lib.linux-armv7l-cpython-311/IT8951
      copying IT8951/interface.py -> build/lib.linux-armv7l-cpython-311/IT8951
      copying IT8951/__init__.py -> build/lib.linux-armv7l-cpython-311/IT8951
      copying IT8951/display.py -> build/lib.linux-armv7l-cpython-311/IT8951
      running build_ext
      building 'IT8951.spi' extension
      creating build/temp.linux-armv7l-cpython-311
      creating build/temp.linux-armv7l-cpython-311/IT8951
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.11 -c IT8951/spi.c -o build/temp.linux-armv7l-cpython-311/IT8951/spi.o
      IT8951/spi.c: In function ‘__Pyx_TraceSetupAndCall’:
      IT8951/spi.c:22082:37: error: invalid use of incomplete typedef ‘PyFrameObject’ {aka ‘struct _frame’}
      22082 |         if (CYTHON_TRACE && (*frame)->f_trace == NULL) {
            |                                     ^~
      IT8951/spi.c:22084:21: error: invalid use of incomplete typedef ‘PyFrameObject’ {aka ‘struct _frame’}
      22084 |             (*frame)->f_trace = Py_None;
            |                     ^~
      IT8951/spi.c:438:62: error: invalid use of incomplete typedef ‘PyFrameObject’ {aka ‘struct _frame’}
        438 |   #define __Pyx_PyFrame_SetLineNumber(frame, lineno)  (frame)->f_lineno = (lineno)
            |                                                              ^~
      IT8951/spi.c:22091:5: note: in expansion of macro ‘__Pyx_PyFrame_SetLineNumber’
      22091 |     __Pyx_PyFrame_SetLineNumber(*frame, firstlineno);
            |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      IT8951/spi.c: In function ‘__Pyx_PyBytes_Equals’:
      IT8951/spi.c:23312:13: warning: ‘ob_shash’ is deprecated [-Wdeprecated-declarations]
[... 117 lines cut ...]
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
IT8951/spi.c: In function ‘__Pyx_PyBytes_Equals’:
IT8951/spi.c:23312:13: warning: ‘ob_shash’ is deprecated [-Wdeprecated-declarations]
23312 |             hash1 = ((PyBytesObject*)s1)->ob_shash;
      |             ^~~~~
In file included from /usr/include/python3.11/bytesobject.h:62,
                 from /usr/include/python3.11/Python.h:50,
                 from IT8951/spi.c:6:
/usr/include/python3.11/cpython/bytesobject.h:7:35: note: declared here
    7 |     Py_DEPRECATED(3.11) Py_hash_t ob_shash;
      |                                   ^~~~~~~~
IT8951/spi.c:23313:13: warning: ‘ob_shash’ is deprecated [-Wdeprecated-declarations]
23313 |             hash2 = ((PyBytesObject*)s2)->ob_shash;
      |             ^~~~~
/usr/include/python3.11/cpython/bytesobject.h:7:35: note: declared here
    7 |     Py_DEPRECATED(3.11) Py_hash_t ob_shash;
      |                                   ^~~~~~~~
IT8951/spi.c: In function ‘__Pyx_AddTraceback’:
IT8951/spi.c:438:62: error: invalid use of incomplete typedef ‘PyFrameObject’ {aka ‘struct _frame’}
  438 |   #define __Pyx_PyFrame_SetLineNumber(frame, lineno)  (frame)->f_lineno = (lineno)
      |                                                              ^~
IT8951/spi.c:24384:5: note: in expansion of macro ‘__Pyx_PyFrame_SetLineNumber’
24384 |     __Pyx_PyFrame_SetLineNumber(py_frame, py_line);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1

GregDMeyer, 2023-10-12:
I think your later issues are due to the need to upgrade the Cython generated files---there is a pull request open for that; unfortunately I have been moving recently so my raspberry pi and epaper display are stuck in moving boxes right now. I should be able to get to them and update in a couple weeks.

For now, can you try running the following commands to install:

```bash
pip install cython~=0.29
export USE_CYTHON=1
pip install ./
```

This should cause the Cython generated files to be regenerated, and compatible with Python 3.11.

By the way, I would try to install as a regular user instead of root, unless you have some crucial need to install as root.

Hoerli1337, 2023-10-13:
Just no hurry, moving is more important than my mistake :)

But now I could install the driver successfully and run the test script.
Now I have a sleeping Tux on the display.
With the stand I can continue.
Thanks for the quick help!

txoof, 2023-10-17:
I'm adding my +1 to this. My project - PaperPi depends heavily on this and I would greatly appreciate any work you can do on making it an easy install for Python 3.11.

If there's any support, testing or anything else I can offer, please let me know.

GregDMeyer, 2023-10-17:
Sounds good. My plan is to switch to `pyproject.toml` and then include Cython in the build requirements, so there should be no longer a need to update things for new versions of Python since the Cython version will track it. I am currently traveling but will be back at the beginning of November, I'll make it a plan to implement this and unpack the box with my EPD in it as soon as I get home!

SirDagen, 2023-10-19:
I really love your software and the assembly I built. 
I do not dare to update my Pi, because of possible complications :)
+1 for an easy install on the current pi image.

GregDMeyer, 2023-11-04:
Just wanted to close this out by saying that I got my pi out of the moving boxes and made the update to use `pyproject.toml`---hopefully everything should work now! See 777d3473ec4ce64eee1326cf8e58dd17e30b8025 for the changes.

closed 2023-11-04 by GregDMeyer

Hoerli1337, 2023-11-05:
Sorry for the late response, unfortunately I had other things to do.
I tried everything again today to make sure that everything runs smoothly.
The installation worked without any problems and I was able to run the test scripts.

Perfect!