GregDMeyer/IT8951 #62: communication with device failed

galshd, 2024-04-01:
I'm using the 7.5inch display, https://www.waveshare.com/wiki/7.5inch_HD_e-Paper_HAT. I've browsed through similar issues (e.g. https://github.com/GregDMeyer/IT8951/issues/18) and tried the following:

- lower spi_hz
- turn spi on and off
- reboot
- check the cables, plug and unplug

However I'm still getting "communication with device failed" when running

```
python3.8 test.py
```

in `IT8951/test/integration`

For what it's worth, the C WaveShare code from https://www.waveshare.com/wiki/7.5inch_HD_e-Paper_HAT works fine.

Is there something I might be missing here? Appreciate any pointers :) thank you!

Nico04, 2024-05-31:
Same here.
Using a Raspberry Pi Zero worked fine, but with my new Pi 5 I have this issue, so sad

GregDMeyer, 2024-06-27:
Please let me know if you have any luck figuring out what is going on---unfortunately since I cannot reproduce the issue I don't have a good way of debugging what is happening. I am certainly open to pull requests if you figure out a fix!

brookstalley, 2024-07-13:
Having the same issue. Here's what I know:

- Does not seem related to running as root / sudo or not. I set up a whole install as root, same problem
- Waveshare demo files work. I've rebooted after running them
- The error is in interface.py:105, where SPI is initialized and apparently the read function just gets zeros back

I'm at a loss, but commenting in the hopes someone solves it.

brookstalley, 2024-07-15:
Dove deeper -- no solution but leaving notes in case anyone else runs across this.

The [example file](https://github.com/waveshareteam/IT8951-ePaper/tree/master/Raspberry/examples) in the waveshare repo works when compiled. I've double-checked everything I can find to align implementations between this repo and that one. I confirmed that all constants for pins and message values are the same. 

There are a couple of differences between the two. The waveshare version:
- Implements reset with high for 200ms, low for 10ms, high for 200ms
- Sets CS low at beginning of write() command and high after write
- Waits for HRDY / ready between sending preamble and payload

I've aligned my local copy of this repro with the waveshare implementations... and still no luck.

brookstalley, 2024-07-16:
While the example file in IT8951-ePaper does work when compiled with the default BCM driver, it DOES NOT work when compiled to use GPIOD (make -j4 LIB=GPIOD). The program starts but it does not successfully control the display at all.

Similarly, manually accessing the SPI ports using pigpiod / pigs also fails the same way: no errors, but all read bytes are returned as zero.

So I'm pretty confident that this issue is not in this IT8951 library at all, but something at the GPIOD level.

pipplo, 2024-08-30:
@brookstalley did you have any luck? I'm having what appears to be the same issue.  Waveshare examples work when using GPIOD but I end up with a 'communication with device failed' error as well.

pipplo, 2024-08-30:
> Dove deeper -- no solution but leaving notes in case anyone else runs across this.
>
> The [example file](https://github.com/waveshareteam/IT8951-ePaper/tree/master/Raspberry/examples) in the waveshare [...]

Do you happen to have these changes available that you can push? I also noticed a few differences in the startup sequence. It looks like gpiod works for me but not for you so maybe  your changes will work on my side?

pipplo, 2024-08-31:
Well a small update, I just tried refreshing with a new image and not installing or following any of the waveshare instructions. The same hardware with the new image works fine. So something leads to an incompatibility [here](https://www.waveshare.com/wiki/6inch_HD_e-Paper_HAT)

🤷

SandrRu, 2024-11-03:
Having the same issue.
With [13.3inch e-Paper Module](https://www.waveshare.com/wiki/13.3inch_e-Paper_HAT)
` line 105, in update_system_info raise RuntimeError("communication with device failed")`

juleaume, 2025-01-10:
I had the same issue with a rasp4. The [Waveshare tutorial](https://www.waveshare.com/wiki/6inch_HD_e-Paper_HAT#Working_with_Raspberry_Pi_.28SPI.29) states to change a value in `/boot/firmware/config.txt`
[image]

Restoring the default value (so `dtparam=spi=on`) solved the issue for me