GregDMeyer/IT8951 #63: Are all IT8951 Boards supported?

DerPicknicker, 2024-04-06:
Hi, 
I already asked here. But I think here is the better place. Some Chinese manufacturers offering a nice 13.3" panel with an controller. They often use an IT8951 controller. 

So are all controller supported besides the wavsehare ones? 

Those boards also supports usb does your driver work via usb as well?

DerPicknicker, 2024-04-06:
See also here : https://github.com/txoof/PaperPi/discussions/157#discussioncomment-9026558

GregDMeyer, 2024-04-11:
Hi!
Unfortunately I don't have any insight into whether this driver would work with those boards. My guess is probably not, because I think WaveShare flashes some custom firmware to their IT8951 boards. But really I don't know, and I don't think there is a good way to find out without someone just getting one of those boards and trying it.
Re: USB, this module is written specifically to use the boards in SPI mode. 
Best of luck with your e-paper endeavors, I hope you find the right hardware for you!